synman/Ender-3-S1-Pro-Firmware #2: Temperature stays at 70 for bed and nozzle after leveling process

DeDaMrAzR, 2023-01-30:
Once leveling is selected and the temps have not reached 70 the screen will default to the home screen (not the leveling screen) and stay there, you have to stop the process and start leveling again. 

Once leveling is done it will remain at 70 for both heaters until the print file is selected and started or power cycle is introduced. 

S1 pro, F4 controller default CR touch location.

synman, 2023-01-30:
I thought this was a fluke when I was doing my own testing but, yeah, I noticed it too.

I'll take a look at this tonight and either fix the logic to work or provide a version that doesn't preheat.

FWIW, commanding via GCode works fine (is primarily how I manage my own ABL too which probably why I missed this after restoring defaults.

```
G28
G29
```

DeDaMrAzR, 2023-01-30:
Confirmed GCode commanding works. 

But it will be better without it :) looking forward to try the new build.

DeDaMrAzR, 2023-01-31:
if it is of any help, Z home from touch screen input triggers this as well...

synman, 2023-01-31:
makes sense sense as the current logic is tied to using the z probe.  Whether it's a G28 or G29, it will check the hotend / bed temperatures.

synman, 2023-01-31:
I have this working locally now and will likely cut a new binary tonight.

DeDaMrAzR, 2023-01-31:
Nice, I'll be happy to test it out. I also got my laser module so I'll test with that as well.

closed 2023-02-01 by synman

ReMimiro, 2023-05-22:
Hi! @synman! Hi everyone! 
I think the issue is back
I use the version 2.1.x-F4-s1pro_abl