synman/Ender-3-S1-Pro-Firmware #3: Feedback needed for "F1" SoC owners

synman, 2023-01-30:
I only have the "F4" board and am unable to validate the "F1" one.

If you have an F1 board, please share your feedback here.

ThomasToka, 2023-02-08:
Today i had a user that tried a build of my version for F4 (he did not know its F4 cause he did not read the readme... He has a F1)

He could start my F4 build on his F1 printer but had several bugs. bed leveling issue, extruder issue.. 

Then i tried to compile with

#default_envs = STM32F103RE_creality 
and
#define MOTHERBOARD BOARD_CREALITY_V24S1_301

compile went fine. but it bricks the installation. screen shows the usual points. LCD is up to date at 1.02

Then i pointed him to your F1 and F4 releases.. The same here. F1 bricked, F4 starts.

His original firmware is  Ender-3S1_Pro_HWv24S1_301_SWV2.0.8.23F1_F103_FDM_LASER.bin .

Any idea whats going on?

I also only have a F4 board. And with that all works great with my builds.

ThomasToka, 2023-02-08:
Ok we just sorted it out.. It is a F4. Bugs on both builds. Yours and mine.. To yours he wrote "His was terrible". Sorry dont want to blame. Both dont work as they should. So false alarm as to F1 Builds.

He just sent me a picture of his stock firmware and it is definitely F4..

synman, 2023-02-18:
going to close this having received feedback elsewhere there are no F1 specific issues.

closed 2023-02-18 by synman