synman/Ender-3-S1-Pro-Firmware #4: Adjust bed size and offsets

synman, 2023-01-30:
Reference:
```
#define NOZZLE_TO_PROBE_OFFSET { -31.5, -41.8, -2.60 }
#define PROBING_MARGIN 40
#define MULTIPLE_PROBING 2
#define EXTRA_PROBING 1
#if ENABLED(ENDER_3S1_PRO)
// The size of the printable area
#define X_BED_SIZE 235
#define Y_BED_SIZE 235
// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -2
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE+20
#define Y_MAX_POS Y_BED_SIZE+6
#define Z_MAX_POS 250
#endif
```

synman, 2023-01-31:
Some of these values don't make sense, mainly increasing the bed size.

I'm inclined to keep the bed dimensions as is and allow for offsets via slicers to gain access to the points beyond them (PrusaSlicer already does this with a negative offset on X to draw its prime line.

I'm taking the x/y probe offsets on good faith.  I have not measured them myself.  Z offset will remain 0.

In place of changing the bed size, I have dropped the PROBING MARGIN to 0.

MULTIPLE PROBING and EXTRA PROBING will remain as is because the idea here is to get a higher quality mesh and more samples == better precision.

ThomasToka, 2023-02-01:
As for the bad size: They had no chance doing it otherwise with the stock firmware cause someone at creality did the bedsize for ender 3 s1 pro simply wrong. So you should start doing it right now as we have the possibilities now. 235mmx235mm is the bedsize not 220mmx220mm. on the z i cut a bit to 250 cause of the cable.

ThomasToka, 2023-02-01:
I first reapected the real physical dimensions. So my settings respect the real stock limits. Relativey precise.
Thats what you should Start with, cause abl depends also on the correct offsets.
Just check where your nozzle really is AT x=0 and y=0 in Relation to the bed..
And if you do it correctly you will see that probe margin 40 is also not bad. Especially respecting y axis on the last probes.

synman, 2023-02-03:
This is getting bumped (and addressed with some tweaks).

The Bed dimensions / mesh size / offset / inset were all contributing to why UBL would not work.... Which I now I have working:)

Yup, someone at Creality screwed up.

ThomasToka, 2023-02-03:
Today i changed my x axis from #define X_MAX_POS X_BED_SIZE+20 to #define X_MAX_POS X_BED_SIZE+10 in order to respect the cable which with 10 has enough space on the right. using 40 as probe margin +10 is absolutely enough.

synman, 2023-02-03:
It's a little different with UBL.  You have to take the full x/y probe offset into account when defining the mesh dimensions.  If you don't the initial `G29 P1` will fail.

synman, 2023-02-03:
this is what I'm currently working through:  https://github.com/MarlinFirmware/Marlin/issues/17494

synman, 2023-02-03:
I have a new build out there for 2.1.2 / 2.1.x-bugfix that should address this

ThomasToka, 2023-02-04:
please just home your x axis. is it really at 0 ? my nozzle is at -2.

And: Using full bed you wont reach y axis with a probe margin of 5. If i remember correctly the probe stands 31.5 mm in the bed towards the front of the printer.. as y back of the printer endstop position is 0. please check this also.

prove me wrong but, i think you do to to complex with your + and - plays with wrong numbers. 

much appreciate your work. r.e.s.p.e.c.t.

synman, 2023-02-04:
Not sure what you're referring to.  You'll need to provide some real references (ideally from one of the 2.1 branches because 2.0.8.is falling behind).

There are two items I'm working to optimize here.

- General bed to nozzle geometry
- Maximize captured mesh for ABL/UBL

ThomasToka, 2023-02-04:
I refer to the real positions in relation to the bed on endstop positions. these are the points to start with a correct measurement.
my ender 3 s1 pro stands on the endstops like i have written:
these are the x and y min positions:
#define X_MIN_POS -2 (nozzle has 2 mm to the x bed left edge)
#define Y_MIN_POS 0 (nozzle is right above the y bed back edge)
next: x axis mesh size is no problem, our problem is y.
if you have your bed at y=0 endstop, it is concludent that having a -31,5 offset to the probe and -1 mm to not hit the endstop.. our minimum possible probe margin would be 32,5mm.. on the y axis back site of the printer... (i choosed here 40mm probe margin, cause id dont like odd numbers.. and 40 with extrapolation is enough..) 
hope you understand where to start with correct messurements.

synman, 2023-02-04:
please reference source, actual current source please, when you do.

ThomasToka, 2023-02-05:
[image]
[image]
[image]

Will comment with explanations to the pictures.

ThomasToka, 2023-02-05:
first picture: you see that the nozzle is 2mm beyond the bed at 0 enstop position. so #define X_MIN_POS -2
second picture: the max position we can reach with y axis is 241
third picture: if we reached 241 max y position our probe -> bed edge offset is around 37mm. so we can reach with probing this 37 mm.. so we have to set the margin around this 37mm.. i choosed 40mm cause i dont like odd numbers here.. 

am i wrong ?

ThomasToka, 2023-02-05:
conclusion: using probe margin 5 will cut y axis on the last row.

lf: x=5mm y=5mm (left front)
... other points
rf: x: 230mm y=5mm (right front)
... other points
lb: x=5mm y=37mm (left back)
... other points
rb: x=230mm y=37mm (right back)

ThomasToka, 2023-02-05:
i corrected it now. cause i initially wrote #define X_MIN_POS -2, which is also wrong. #define Y_MIN_POS 0 is perfect. But
#define Y_MIN_POS 0 not.
it should be #define Y_MIN_POS -2

ThomasToka, 2023-02-05:
hm i will recheck #define X_MIN_POS also when my printer is ready.. i had really #define X_MIN_POS -2 and #define Y_MIN_POS 0 in my config.. strange. maybe all was hot air or not. but it should be correct. will report later again when i changed it to #define X_MIN_POS 0 and #define Y_MIN_POS -2. then i see if #define X_MIN_POS -2 was also correct, as i noticed that my head goes back on x after hitting endstop 1 or 2mm. which it did not at y.

synman, 2023-02-05:
Believe it or not, the stock MIN positions are pretty darn close.

Regarding Y, you have to contend with the proximity of the bed wires block.  Many people have a custom cooler installed and if you push the Y MAX too far we run the risk of a cooler hitting the block.

ThomasToka, 2023-02-05:
Sorry But you are wrong.

I have made two short videos of your build and my build.

Hope i can upload here.

ThomasToka, 2023-02-05:
Hm does not work. Will sent you on Facebook.

ThomasToka, 2023-02-05:
after all tests my perfect numbers now are:

  // The size of the printable area
  #define X_BED_SIZE 235
  #define Y_BED_SIZE 235

  // Travel limits (mm) after homing, corresponding to endstop positions.
  #define X_MIN_POS -0.8
  #define Y_MIN_POS -2
  #define Z_MIN_POS 0
  #define X_MAX_POS X_BED_SIZE (no need to go with the nozzle more to the left if X_MIN_POS matches)
  #define Y_MAX_POS Y_BED_SIZE+4 (ends on 239, right before the end if Y_MIN_POS matches)
  #define Z_MAX_POS 250
  
  and to have a square mesh:
  #define PROBING_MARGIN 40

closed 2023-02-06 by synman