synman/Ender-3-S1-Pro-Firmware #7: abl offcenter

IAGITH, 2023-02-01:
will start by saying i have the X-46.5 Y0 bracket for my cr touch. when doing abl all points are on the left side of the bed. i assume this is caused by the right movement limit. is there a way to make it go further right? it can physically do it.

synman, 2023-02-01:
What are you current offsets.  Let's start with seeing if M851 is showing what you expect.

IAGITH, 2023-02-02:
> What are you current offsets? Let's start with seeing if M851 is showing what you expect.

I used m851 to set offset to x-46.5  y0 z0. It just works for me, but even if the home position probes right on center of build surface the mesh grid is gets really close to the left corner is not enough on the right.

synman, 2023-02-03:
I wouldn't change the offsets to accommodate the geometry error.  The entire bed map is shifted so what appears off center is actually what the machine knows as center.

Changing the offsets will throw off your mesh....

I have a fix for this.  I just need to commit it.

synman, 2023-02-03:
This is related to #4

IAGITH, 2023-02-04:
> I wouldn't change the offsets to accommodate the geometry error. The entire bed map is shifted so what appears off center is actually what the machine knows as center.
> 
> Changing the offsets will throw off your mesh....
> 
> --- EDIT ---
> 
> Completely understand if you tweaked the offset due to having an alternate location for your probe. You may need a custom build.
> 
> I have a fix for this. I just need to commit it.

I installed the oem bracket cause i thought maybe that stopped my abl from working on your latest release. Didn’t fix it so i went back to the one that was working. *Wednesday * with eom cr touch location mesh still probed 10mm to the left end of bed, but 25mm on the right (this happened also with creality oficial firmware) but also my mesh was a mess xD, so once i get time I’ll go back to y0 bracket since it gave me incredible results. Looking forward to more releases :D

synman, 2023-02-04:
I fear my builds as is will continue to show your stuff off center if you modify the X/Y offsets.

However, it's pretty easy to build your own clone. Check out the issue I linked here (and the one linked there) for more information about how I ultimately solved this

Solved at least until a better solution surfaces.

synman, 2023-02-28:
closed due to inactivity

closed 2023-02-28 by synman