synman/Ender-3-S1-Pro-Firmware #8: Laser mode

IAGITH, 2023-02-01:
Laser switch button is just blank and does not switch to laser mode, but does make a sound when pressed.

synman, 2023-02-01:
Which build is this on (2.1.2 or 2.0.8)?   I do not have a laser to test and am pretty sure laser will be non-functional on 2.1.2.  It may work on 2.0.8.

IAGITH, 2023-02-02:
> 

I installed 2.1 is there any benefits to 2.1? Since linear advance is not a possibility anyway.

synman, 2023-02-02:
There are 1000s of changes / fixes / enhancements to Marlin between 2.0.8 and 2.1.2.  All of my focus right now is building upon the 2.1.2 branches since I've got it synced up with the core Marlin repo.

I'm going to mark this as an enhancement and try to enable to some of the functionality.  My ability will be limited though as I do not have an actual Creality laser.  I should be able to get it to switch modes though.  What happens after that will be untested.

synman, 2023-02-03:
I have confirmed you can switch to laser mode on the 2.0.8 build.

Will look at addressing for 2.1.2 in the coming days.

synman, 2023-02-23:
will be taking this on for bugfix-2.1.x next.  I'm thinking once I have the laser working in 2.1.x that I'll be retiring / removing the 2.1.2 and 2.0.8 ones.

synman, 2023-02-28:
I have a clean compile for bugfix-2.1.x ... man, there were a ton of changes to forward port Creality's laser logic.

IAGITH, 2023-03-01:
> I have a clean compile for bugfix-2.1.x ... man, there were a ton of changes to forward port Creality's laser logic. 20 source / header files and about 800 lines of code to get this thing clean.

nice. I will give it a go.

synman, 2023-03-01:
fun fact:  the laser uses the heat block fan pin for pwm

this has been delivered untested as I do not have a laser.

IAGITH, 2023-03-01:
> fun fact: the laser uses the heat block fan pin for pwm
> 
> this has been delivered untested as I do not have a laser.

I’ll get the laser going once i get home. How does it use the parts cooling fan? It is only connected to the lightbar plug.. 🤔

synman, 2023-03-01:
I have no idea but I had to do some trickery within the pins files to get to compile.

It appears Creality commented out the Sanity checks for it rather than implement it properly.  I wonder if what is labeled FAN1 is actually the light bar.  I did notice previously there's a PWM connector around there.

IAGITH, 2023-03-01:
> I have no idea but I had to do some trickery within the pins files to get to compile.
>
> It appears Creality commented out the Sanity checks for it rather than implement it properly. I wonder if what [...]

Okay. Installed it and  switched to laser no problem. Laser starts off and finishes off(was a bit worried that I wouldn’t shut off). Only issue is that it does not run “range” usually when lasering before engrave you can check the actual size and position the laser will engrave. The button was there, but it will just not run it, so i just click start and hoped for the best 🤣 (ended up off center) 
[image]

synman, 2023-03-01:
wow, I have to admit, I would not have bet on it working  Got super lucky given the extent of changes I had to make.  

So it comes down to basically a bounding box menu not working?   Can you share a picture of that function so I have something to go off of?  I'll see about locating it within the screen handler.  maybe there's something stupid missing.

synman, 2023-03-01:
So now I'm wondering what benefit laser gets from stuff like LA, IS, and Backlash Compensation.   This is a major leap from the Creality 2.0.8 code... I just don't know if it all translates to laser.

synman, 2023-03-01:
> I’ll get the laser going once i get home. How does it use the parts cooling fan? It is only connected to the lightbar plug.. 🤔

The reference that me worried was for the extruder / heat block fan.... not the parts cooling fan.   I haven't seen how the laser physically attaches.  Is it really getting all of its power and the PWM signal just from that little wire that runs the light?  I know it's 3 pin so it's certainly feasible... just really small gauge wires for running the necessary current on the power line.

synman, 2023-03-01:
I'm wondering if this is what we're looking for
```
      }else if(recdat.data[0] == 4) //跑边框
      {
        HMI_Area_Move();
```

跑边框 translates to "run border"

IAGITH, 2023-03-02:
> wow, I have to admit, I would not have bet on it working. Got super lucky given the extent of changes I had to make with zero ability to test it.
> 
> So it comes down to basically a bounding box menu not working? Can you share a picture of that function so I have something to go off of? I'll see about locating it within the screen handler. maybe there's something stupid missing.

i engraved right before sleep. Anyway here is picture. Yes it will flat out ignore the command, but do make a sound when pressed. 
the only wire going to the laser is the one that connects to the light bar connector mine is only 1.6W laser, so it doesn’t take much at all 
[image]

synman, 2023-03-02:
Does the axis move button work?

IAGITH, 2023-03-02:
> Does the axis move button work?

Yes axis moves before and during laser. works like z offset and i can lower if it’s too high after it already started

IAGITH, 2023-03-02:
> So now I'm wondering what benefit laser gets from stuff like LA, IS, and Backlash Compensation. This is a major leap from the Creality 2.0.8 code... I just don't know if it all translates to laser.

LA is the extruder i believe.. input shaping is the anti vibration(i think), and backlash is from reversing a motor.. laser just moves in a straight line then just does the same backwards.. it isn’t crazy fast cause it’s got to burn something.. speed is constant as well, so i doubt any benefit will come from any of those.. i doubt it even uses them 🤔 all the option on slicer is 1500 mm/min strength of laser and what style you want to burn.. gray scale/ different gray scale/ black and white

synman, 2023-03-02:
I'm used to grbl when it comes to laser / CNC and some of these Marlin features are built into grbl (like backlash I believe).

I guess LA would be limited to extrusions, makes sense.

I'm thinking I'm going to need to insert some debug statements into this print range thing.  Are you willing to try it and just as important, do you have an external sender we can use to capture the output?

IAGITH, 2023-03-02:
> I'm used to grbl when it comes to laser / CNC and some of these Marlin features are built into grbl (like backlash I believe).
>
> I guess LA would be limited to extrusions, makes sense.
>
> I'm thinking [...]

I’ll try it.. but I don’t know what you mean by external sender.. like pronterface?

synman, 2023-03-02:
Exactly.  There will be debug statements written out the serial interface I'll need to review to get a hint for what's not working.

IAGITH, 2023-03-02:
> Exactly. There will be debug statements written out the serial interface I'll need to review to get a hint for what's not working.

Yeah i can connect pronterface and do a quick laser. Just copy text and send it to you :)… i have stuff going on this morning, but probably will do at night or whenever you have it

synman, 2023-03-02:
I'll try to get a build built just for you.  F4 chip/UBL_100 ?

IAGITH, 2023-03-02:
> I'll try to get a build built just for you. F4 chip/UBL_100 ?

F4 chip but I’m using abl

synman, 2023-03-03:
working on adding the debugging lines now and will get that build out there shortly.

synman, 2023-03-03:
[F4-Pro-ABL-Laser-DEBUG.zip](https://github.com/synman/Ender-3-S1-Pro-Firmware/files/10878041/F4-Pro-ABL-Laser-DEBUG.zip)

There should be some good data that comes from this.  Please capture everything logged starting no later than after you select the gcode file you're looking to run.

IAGITH, 2023-03-03:
> [F4-Pro-ABL-Laser-DEBUG.zip](https://github.com/synman/Ender-3-S1-Pro-Firmware/files/10878041/F4-Pro-ABL-Laser-DEBUG.zip)
> 
> There should be some good data that comes from this. Please capture everything logged starting no later than after you select the gcode file you're looking to run and until after you select the range option and nothing happens. Feel free to hit the range button a couple times too.
So button did react but only move laser a bit to the right. Then i homed all axis and bit range again. Laser again moved a bit to the right(not running range) 
Tried a third time and same result(this last time didn’t fit in my camera. My pronterface laptop is not network connected)
Print Pause off
Connecting...
Printer is now online.
Checkkey=53 recdat.data 0]= 1 echo:Now fresh file: ender-~1.go
File opened: ender-~1.go Size: 2956951
File selected
//action:notification Ender-3 S1 Laser_alice cartoon.gcode
Checkkey=57 recdat.data 0j=4 caling HMI_Area_Move
HMI Area Move Start
HMI_Area_Move 1st move
RTS_line_to_current axis= (1] feedrate= [3000.0000]
RTS_line_to_current = (0.0000] y= (0.0000]
HMI Area Move laser on (weak)
HMI_Area_Move 2nd move
RTS_line_to_current axis= [1] feedrate= [3000.0000)
RTS line to current = [0.0000] y= (0.0000)
HMI_Area_Move 3rd move
RTS_line to_current axis= [0] feedrate= (3000.0000]
RTS_line_to_current x= [0.0000] y= (0.0000]
HMI_Area_Move 4th move
RTS_line_to_current axis= (1] feedrate= (3000.0000]
RTS_line_ to_current = [0.0000] y= [0.0000]
HMI_Area_Move 5th move
RTS_line_ to_current axis= (0] feedrate= (3000.0000]
RTS_line_to_current x= [0.0000] y= (0.0000]
HMI_Area_Move laser off
HMI_Area_Move 6th move
RIS_line_to_current axis= (0] feedrate= [3000.0000]
RTS_line_to_current x= (0.0000] y= (0.0000]
Checkkey= 57 recdat.data[0]=2
Checkkey=60 recdat.data[0]=5 echo:enqueueing "G28 XY F3000* echo:enqueueing "G1 XO Y10 F3000*
Checkkey=60 recdat.data(0]=4
Checkkey=57 recdat.data[0]=4 caling HM_Area_Move
HMI_Area_Move Start
HMI_Area_Move 1st move
RTS_line_to_current axis= [1] feedrate= [3000.0000]
RTS_line_to_current = (0.0000] y= [10.0000]
HMI_Area_Move laser on (weak)
HMI_Area _Move 2nd move
RTS_line_to_current axis= (1] feedrate= [3000.0000]
RTS_line_to_current x= [0.0000] y= [10.0000]
HMI_Area _Move 3rd move
RTS_line_to_current axis= (0] feedrate= [3000.0000]
RTS_line_to_current ×= (0.0000] y= [10.0000]
HMI_Area _Move 4th move
RIS_line_to_current axis= (1] feedrate= (3000.0000]
RIS_line_to_current x= (0.0000] y= [10.0000]
HMI_Area_Move 5th move
RIS_line_to_current axis= (0] feedrate= [3000.0000)
RTS_line to_current ×= [0.0000] y= [10.0000]
HMI_Area_Move laser off
HMI_Area_Move 6th move
RIS_line,_ to_current axis= (0] feedrate= [3000.0000]
RTS_line_to_current = [0.0000] y= [10.0000]

synman, 2023-03-03:
This is extremely helpful!  Unfortunately it means I need to look at an even deeper Creality routine where it computes the bounding box size.

I can see all it did was move Y 10mm.

Please attach your GCode file too.  I think I know where this is headed as I have a bit of experience computing bounding box dimensions in the grbl world (see my better grbl support plugin for Octoprint for reference).

synman, 2023-03-03:
laser_device.get_laser_range is suspected root cause

IAGITH, 2023-03-03:
> This is extremely helpful! Unfortunately it means I need to look at an even deeper Creality routine where it computes the bounding box size.
> 
> I can see all it did was move Y 10mm but I need to correlate that to menu ordinal referenced. I'm certain it's the move command you referenced.

> Please attach your GCode file too. I think I know where this is headed as I have a bit of experience computing bounding box dimensions in the grbl world (see my better grbl support plugin for Octoprint for reference).
----------------------
> here is the gcode (had to zip it to send it) 
[Ender-3 S1 Laser_alice cartoon.zip](https://github.com/synman/Ender-3-S1-Pro-Firmware/files/10882797/Ender-3.S1.Laser_alice.cartoon.zip)

synman, 2023-03-03:
I'm going to spin this off as its own bug too.  You'll see me move future updates to that issue once I have it created.

synman, 2023-03-03:
Scratch that... I have confirmed root cause.   I'm missing some merge elements from the 2.0.8 Creality code drop.    I went to look for their code that computes the bounding box size and it was completely absent from my merged code.... and sure enough, it is there plain as day in the 2.0.8 code.

Assuming I don't have to completely rewrite this to work within 2.1.x I should have it fixed tonight.

synman, 2023-03-03:
I like Creality less and less every day.  This is such a pitiful hack.  I'm actually taken back by it.

So they don't actually calculate the bounding box size.  they are merely sniffing the top portion of the gcode file reading comments looking for some values that map back to some variables they have defined in the screen handler.
```
;Header Start
;estimated_time(s): 2675.933    
;MAXX: 156.67
;MAXY: 155.67
;MINX: 48.67
;MINY: 61.33
;Header End
```
This is such a horrible hack IMHO.  They are locking you into their "slicer" with this hack because I guarantee you you're not going to get this type of header written out with a tool such as Lightburn.

It pains me to share that I have added this code.   Yet again, another perfect example of why the Marlin team will never accept any code from Creality.

IAGITH, 2023-03-04:
> I like Creality less and less every day. This is such a pitiful hack. I'm actually taken back by it.
>
> So they don't actually calculate the bounding box size. they are merely sniffing the top portion [...]

I see.. i thought from the start i could only use creality print 🤣 don’t know much about engraving, but yeah not cool to use open source to be greedy like that. Then again the entire machine is a bunch of proprietary stuff, so I’m not surprised

synman, 2023-03-04:
The version of the binary I shared here: https://github.com/synman/Ender-3-S1-Pro-Firmware/discussions/38#discussioncomment-5200073 has the laser bounding box fix in it if you don't want to wait for my next full build.

It'll probably be a day or two until I do another full release as there are a couple other things I want to address.

synman, 2023-03-04:
> I see.. i thought from the start i could only use creality print 🤣 don’t know much about engraving, but yeah not cool to use open source to be greedy like that. Then again the entire machine is a [...]

This also tells me you are bound to SD based engraving if you want to use these "features".  Ah man, it's a shame you don't have a grbl based laser. Nice for the perfect fix for you.

thomazff, 2023-05-12:
Does you firmware has Laser_power_inline eneabled? Trying to use this feature on Lightburn, but I think creality kept this disabled (as i believe marlin has this as default).