synman/Ender-3-S1-Pro-Firmware #11: Linear Advance Support

synman, 2023-02-02:
Placeholder for further refinement

ThomasToka, 2023-02-02:
K 0.00 as default or a reasonable setting for the s1 pro?

I have mine at 0.06 after calibration.

InD1ggo2, 2023-02-05:
Does the Linear Advance function work in your firmware now? I tried to print a calibration model to calculate the coefficient "K", but there is no difference between the coefficients in the printed model.

synman, 2023-02-05:
haven't started on it yet.  will update this issue once I do.

synman, 2023-02-06:
I'm getting conflicting information on whether our stepper drivers are even compatible.

I may proceed with an additional variant (already have two -- UBL vs ABL) but the last thing I want to do is introduce a feature that could fry a board / driver.

synman, 2023-02-06:
adding some reference material here for guidance.

https://all3dp.com/2/linear-advance-cura-marlin/

I'm tempted to enable it with a 0 K value.  I believe it is basically anemic in this manner and if you want to tweak it, you can do it via gcode and m500.

synman, 2023-02-06:
First point of contention is that our steppers are in `STANDALONE` mode:
```
#define X_DRIVER_TYPE  TMC2208_STANDALONE
#define Y_DRIVER_TYPE  TMC2208_STANDALONE
#define Z_DRIVER_TYPE  TMC2208_STANDALONE
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
//#define Z4_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2208_STANDALONE
//#define E1_DRIVER_TYPE A4988
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988
//#define E6_DRIVER_TYPE A4988
//#define E7_DRIVER_TYPE A4988
```

synman, 2023-02-06:
video where it's called out as an issue for us:  https://www.youtube.com/watch?v=n3yK0lJ8TWM

he doesn't elaborate, just says it is a "NO GO".

synman, 2023-02-06:
interesting comments which match our use case:

> [Nova Leary](https://www.youtube.com/channel/UCYY0w-ctsEfJ50ayouFHoYA)
> [3 years ago (edited)](https://www.youtube.com/watch?v=n3yK0lJ8TWM&lc=UgxYJLLyQKrMUyaErU54AaABAg)
> Thanks for the video! I wanted to add in some with reguarding the TMC2208s. It is possible to use Linear Advance on TMC2208s so as long as you disable StealthChop and enable Square Wave Stepping.
> 
> 
> Edit:: I can confirm that Linear Advance works on TMC2208 drivers on the extruder! You have to ensure that Stealthchop is disabled on the extruder and have square wave stepping enabled

synman, 2023-02-06:
and it is disabled by default:
```
 /**
   * Beta feature!
   * Create a 50/50 square wave step pulse optimal for stepper drivers.
   */
  //#define SQUARE_WAVE_STEPPING
```

synman, 2023-02-06:
this is quite a little rabbit trail...

https://reprap.org/forum/read.php?415,863853

says we should enable SpreadCycle for the extruder...  SQUARE_WAVE_STEPPING does not work in standalone mode apparently.

synman, 2023-02-06:
Another option referenced is MINIMUM_STEPPER_PULSE.   It defaults to 0 on TMC.   Guidance is for it to be 1 or greater.

```
  #elif HAS_TRINAMIC_CONFIG || HAS_TRINAMIC_STANDALONE
    #define MINIMUM_STEPPER_PULSE 0
```

synman, 2023-02-06:
There is no way to configure StealthChop vs SpreadCycle in standalone mode and the default is StealthChop.  It is possible setting `MINIMUM_STEPPER_PULSE` will be enough

synman, 2023-02-06:
Found this today

https://github.com/MarlinFirmware/Marlin/pull/24533

It is only in bugfix-2.1.x 

Based on this, LA is off limits for 2.0.8 and 2.1.2...

synman, 2023-02-07:
> K 0.00 as default or a reasonable setting for the s1 pro?
> 
> I have mine at 0.06 after calibration.

0.06 is coming up as the sweet spot on my machine as well.  I had to dramatically change the scale on the calibration generator to make it useful.

ThomasToka, 2023-02-07:
cool you sorted it out. yes i think it was so long below the radar as you have to make the testrange very fine to see it works correctly. thank you for the link to #24533 . did not know the fact that there was a rewrite.. but have seen that it works months ago as my ender3pro without S1 has 2208 drivers. and it worked there fine after calibration.

cheers.

synman, 2023-02-07:
This is now enabled in the bugfix-2.1.x builds and I am only going to enable it in them.

I default the K value to 0 since you pretty much have to tune linear advance on a material / filament type / brand basis.

k values of .06 (and even .07) appear to be stable, however I've been unable to evidence any visual improvement in my benchy benchmarks.