synman/Ender-3-S1-Pro-Firmware #18: CHOPPER_TIMING CHOPPER_DEFAULT 12V or 24V

ThomasToka, 2023-02-07:
Its not  a bug but more a question:

`#define CHOPPER_TIMING CHOPPER_DEFAULT_12V        // All axes (override below)`

or

`#define CHOPPER_TIMING CHOPPER_DEFAULT_24V        // All axes (override below)`

Sorry if it is a dumb question. I know that it should work with both as it is a constant power.. 

But i am curious what is better here.. And what is see Ender 3 S1 Pro is also a 24V machine..

Reference to a pull for Ender 3
https://github.com/MarlinFirmware/Marlin/pull/13565

synman, 2023-02-07:
24V makes the most sense to me... and it should already be in both the 2.1.2 and bugfix-2.1.x branches as it goes all the way to the old bugfix-2.0.x branch.... odd that it isn't....

I wonder if this came over from the merge I did from the Creality 2.0.8 repo.... checking now.

synman, 2023-02-07:
could very well be a bug too :)

synman, 2023-02-07:
weird....

https://github.com/CrealityOfficial/Ender-3S1/blob/69404a775d0d3c55bd59c62eac8c1dc6b5a9db6a/Marlin/Configuration_adv.h#L2739

I know Creality is bad, but are they really that bad?   If you're playing with the 24V setting, please let me know your results.

ThomasToka, 2023-02-07:
It works also. I had it in my initial build. That with the 12864 display in the old days before the official release.. But then changed to the "bad setting" to follow "official release".. lol

For me no problem with 24V.

synman, 2023-02-07:
ok, I just looked a little deeper....  it doesn't matter for a stock machine.    That config value is nested within HAS_TRINAMIC_CONFIG and since we are in STANDALONE mode the voltage is hardwired on the PCB.

closed 2023-02-09 by synman