synman/Ender-3-S1-Pro-Firmware #20: It seems motor steps are skipped while pushing touch z-offset fast

ThomasToka, 2023-02-08:
**Describe the bug**
I sorted and reproduced the bug my user had with our builds.
It seems that the touch skips z-offset steps on the motors but records them as set.
yes motor steps, not what the touch shows, that seems to notice and records every push.
for example if you pushed 10 times fast it records all 10 steps for the numbers shown but the stepper just does lets say 8 steps.
I reproduced this on my printer.

**To Reproduce**
1.) note your working z-offset, mine is for example -1.74mm atm
2.) load firmware and reset
3.) go to aux.level leveling to point one
4.) try set your offset to the visible gap you had with -1.74mm
5.) you will notice that you need to enter way more to reach the former -1.74mm position. in my case it was -2.14
6.) recorded is -2.14, real is -1.74mm
7.) do auto bedleveling
8.) start print
9.) you will notice that nozzle is too close to bed.. when pushing here to fast.. it also skips some motor steps but records.. so that what you set is incosistent, as one tries to push fast to up nozzle fast as it grinds.

other wise it works if you do:

1.) note your working z-offset, mine is for example -1.74mm atm
2.) load firmware and reset
3.) set M851 Z-1.74 as offset, M500
4.) do auto bedleveling
5.) start print

**Expected behavior**
Consisntant recording steps and executing motor steps.

**Screenshots**
can deliver if needed.

**Desktop (please complete the following information):**
not needed

**Smartphone (please complete the following information):**
not needed

**Additional context**
Hope you can reproduce, i could.

synman, 2023-02-08:
My guess is the stock firmware will do this too.  I can likely hack it with a half second sleep.

ThomasToka, 2023-02-08:
i like 0.01 microstepping. it is absolutely needed for finetuning z-offset.
possible solutions could be
- fix code if possible
- if not possible maybe add additional + and - buttons for 0.1 babystepping on the z offset site, this would decrease the number pushes needed to reach the needed offset. 1,74 would then be only 17(0,1)+4(0,01) pushes. with 0.01 it would be 174 pushes needed..
just my thoughts.. hope it helps..

ThomasToka, 2023-02-08:
> My guess is the stock firmware will do this too. I can likely hack it with a half second sleep.

yeah but it has 0.05 so five time less pushes.. 174 or 34 is markable here..

ThomasToka, 2023-02-08:
half a second may be too much i think. i would say just make some test with different values.. if say me where i can also do some testing and we can exchange results.. so we need #maxtouchpushspeed now. lol

synman, 2023-02-09:
I was gonna start with a 200ms sleep

synman, 2023-02-09:
> i like 0.01 microstepping. it is absolutely needed for finetuning z-offset. possible solutions could be
>
> * fix code if possible
> * if not possible maybe add additional + and - buttons for 0.1 [...]

we have zero control over the actual touchscreen at this time.  We must use what Creality has made available until Creality releases the source or @Pethical is able to move his rewrite forward (https://github.com/Pethical/Ender-3-S1-Pro-Screen)

so yeah... the best we can do is introduce something to slow down the touch events.

synman, 2023-02-09:
this is fixed in the latest builds (I'm running them now).

closed 2023-02-09 by synman

ThomasToka, 2023-02-09:
just to give you a feedback: works like intended. i really hammerd my z-offset hard. no more skips. well done.

synman, 2023-02-09:
yup.  If you look at the approach I took, I completely removed timing / synchronization from the equation by relying on M290 to do the work for us.

A cool side effect of this is that the already integrated hooks between M290 events and the LCD keeps everything in sync.