synman/Ender-3-S1-Pro-Firmware #26: [BUG] MANUALL_BED_LEVEING_5POSITION int

ThomasToka, 2023-02-09:
### Bug Description

in order to be more precise with leveling points i would like to change MANUALL_BED_LEVEING_5POSITION in config

from:
#define MANUALL_BED_LEVEING_5POSITION {110,110, 30,30, 190,30, 190,190, 30,190}
to:
#define MANUALL_BED_LEVEING_5POSITION {117.5,117.5, 40,40, 195,40, 195,195, 40,195}

### Configuration Files

applies to all in 2.1.x-bugfix

### Steps to Reproduce

change like i have written and compile

**Expected behavior:**

working binary

**Actual behavior:**
`Compiling .pio\build\STM32F401RC_creality\src\src\lcd\extui\dgus\origin\DGUSScreenHandler.cpp.o
In file included from Marlin\src\lcd\e3v2\creality\../../../inc/MarlinConfigPre.h:39,
                 from Marlin\src\lcd\e3v2\creality\../../../inc/MarlinConfig.h:28,
                 from Marlin\src\lcd\e3v2\creality\lcd_rts.h:7,
                 from Marlin\src\lcd\e3v2\creality\lcd_rts.cpp:6:
Marlin\src\lcd\e3v2\creality\../../../inc/../../Configuration.h:2092:85: error: narrowing conversion of '1.175e+2' from 'double' to 'int' [-Wnarrowing]
 2092 |   #define MANUALL_BED_LEVEING_5POSITION {117.5,117.5, 40,40, 195,40, 195,195, 40,195}
      |                                                                                     ^
Marlin\src\lcd\e3v2\creality\lcd_rts.cpp:194:42: note: in expansion of macro 'MANUALL_BED_LEVEING_5POSITION'
  194 | const int manual_level_5position[5][2] = MANUALL_BED_LEVEING_5POSITION;
      |                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\e3v2\creality\../../../inc/../../Configuration.h:2092:85: error: narrowing conversion of '1.175e+2' from 'double' to 'int' [-Wnarrowing]
 2092 |   #define MANUALL_BED_LEVEING_5POSITION {117.5,117.5, 40,40, 195,40, 195,195, 40,195}
      |                                                                                     ^
Marlin\src\lcd\e3v2\creality\lcd_rts.cpp:194:42: note: in expansion of macro 'MANUALL_BED_LEVEING_5POSITION'
  194 | const int manual_level_5position[5][2] = MANUALL_BED_LEVEING_5POSITION;
      |                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pio\build\STM32F401RC_creality\src\src\lcd\extui\dgus_reloaded\DGUSDisplay.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\lcd\extui\dgus_reloaded\DGUSRxHandler.cpp.o
*** [.pio\build\STM32F401RC_creality\src\src\lcd\e3v2\creality\lcd_rts.cpp.o] Error 1
`

synman, 2023-02-09:
I don't think this is used in any of the active variants.  I'm also unimpressed with Creality's inability to spell words properly.

synman, 2023-02-09:
Ok, so I see how this works.  it's feeding the LCD the origin points for the "aux" leveling menu.   Since we are using the stock values, this isn't necessarily a bug as it is working as Creality intended.

However, I do think we can clean it up and make some adjustments to meet our needs.

ThomasToka, 2023-02-09:
i just try to make everything a bit better for all of us as we can do it now. thank you for your understanding.

ThomasToka, 2023-02-09:
btw: #define MANUAL**L**_BED_LEVEING_5POSITION shouild be #define MANUAL_BED_LEVELING_5POSITION
maybe you can fix this just for the sake of "my internal monk" also if follow this variable through code..

synman, 2023-02-09:
It's also unclear to me what you did above there to cause compile time errors.   The code builds fine in its current state.  I'm gathering it is just the values you don't like in your never ending quest to center a grid absent a gutter with a 40mm probe offset to contend with.

jklap, 2023-02-09:
> I'm gathering it is just the values you don't like in your never ending quest to center a grid absent a gutter with a 40mm probe offset to contend with.

LOL. There really needs to be a laugh emoji for comments

ThomasToka, 2023-02-09:
> It's also unclear to me what you did above there to cause compile time errors. The code builds fine in its current state. I'm gathering it is just the values you don't like in your never ending quest to center a grid absent a gutter with a 40mm probe offset to contend with.

it doesnt matter what my probe margin is.
It does matter that the bed is 235x235 and half of it is 117.5. creality did it uneven in the half :-( not my fault. lol
and point 1 should be in the middle. thats all.

synman, 2023-02-09:
@ThomasToka ,

Why don't you just create a fork of the bugfix-2.1.x branch, add in a variant called "nogutter" or similar and submit a PR to bring it back into the mainstream fold? 

Doing so will give you all the freedom you want to tweak the "nogutter" variant however you like and, assuming your PRs are clean, I can just accept them while I'm focusing on items I consider more critical.

If you do this, please limit it to just the S1 Pro builds, unless you have a baseline with the S1 Plus to work with too.

A good place to start if you're not following me is here:  https://github.com/synman/Marlin/blob/bugfix-2.1.x/build-all-ender.sh

and here (a good F4 chip example): https://github.com/synman/Marlin/blob/bugfix-2.1.x/ini/stm32f4.ini#L691