synman/Ender-3-S1-Pro-Firmware #30: Invalid mesh

NicolasZorzettig, 2023-02-12:
**Describe the bug**
I use Pronterface to see what the printer is doing. After the calibration on 100 points has finished, only 97 points are saved. This results in a non valid mesh.

**To Reproduce**
Steps to reproduce the behavior:
1. Factory reset (I tried to do an automatic bed levelling two consecutive times, the second time I didn't do a factory reset)
2. Set temperatures manually (I use 240/80)
3. Wait for the temperatures to reach the set values
4. Launch the automatic bed levelling (I even tried to do a manually bed levelling before this step through the auxiliary bed levelling function)

**Screenshots**
[image]
[image]

**Printer/Firmware:**
- Printer: S1 Pro
- Firmware: STM32F103RE_creality_s1pro_ubl100
- Screen: V1.0.3

**Additional context**
The followings are probably bugs caused by the screen version because I'm using the February version. I will downgrade to the November version and if they persist I will  wrote another report:
- After the calibration has completed, the printer keeps giving me "echo:busy: paused for user" and the only way to stop it is to lauch a M108 command (going back to the home screen isn't enough).
- After the calibration has completed, if I go to the auxiliary levelling window the printer stops responding. I can't move on the 5 points and the return button on the upper-left corner of the screen doesn't work.

synman, 2023-02-12:
Interesting.  I have not touched the Feb code drop yet so it is possible there is some subtle difference there, but I wouldn't bet on it just yet.

You definitely have to do a factory reset when first coming over to the UBL firmware and with any subsequent firmware updates you flash.  So long as you use the same mesh size (_ubl100 vs _ubl25) for future flashes, your previously generated meshes will survive the below procedure.
```
M502 ; factory reset
M500 ; save defaults
M501 ; reload settings (for good measure)
```
Here's the workflow I run for UBL (I actually have the 100 point grid active myself):
```
G28 XYZ ; home xyz
G29 P1 T ; generate new UBL mesh in slot 1
... it will likely fail on one or two points and finish with an incomplete mesh ...
G29 P1 C T ; continue the P1 mesh generation -- it will probe the previously missed points
G29 P3 ; fill gaps but is likely unnecessary because we can probe every point of the mesh
G29 S1 ; save your new mesh to slot 1
```
This gcode ^^^ is wired into the auto bed leveling logic inside the screen handler.   In other words, this is the actual procedure the LCD follows when you generate a mesh from it.

You can recall and use your mesh later (slicer gcode start script) using this:
```
G28 ; home all axis
;
M420 L1 V ; use saved mesh
G29 J2 T; shift mesh (4 corners)
M420 S1 V; activate mesh
```
Anyway, if you are not following this procedure, try it and let me know how things go.   I'm also very interested to know if you tie this to the February screen firmware drop.

We can continue this here https://github.com/synman/Ender-3-S1-Pro-Firmware/discussions/23 if we rule out a firmware issue / bug.

synman, 2023-02-12:
I just noticed... you're on the F1 chip?

> STM32F103RE_creality_s1pro_ubl100

You're the first person I've found using this board.  I doubt it is related but good to know there are actually people out there with the F1 chip.  I was starting to wonder.

NicolasZorzettig, 2023-02-12:
I did the M502 -> M500 -> M501 commands just after upgrading the to the 2.1.2 firmware from the factory one.

> I just noticed... you're on the F1 chip?

Yes, I bought the printer shortly after it came out when it still was built with the F103 chip.

> Here's the workflow I run for UBL (I actually have the 100 point grid active myself):
> 
> ```
> G28 XYZ ; home xyz
> G29 P1 T ; generate new UBL mesh 
> ... it will likely fail on one or two points and finish with an incomplete mesh ...
> G29 P1 C T ; continue the P1 mesh generation -- it will probe the previously missed points
> G29 P3 ; fill gaps but is likely unnecessary because we can probe every point of the mesh
> G29 S1 ; save your new mesh to slot 1
> ```

Thank you. I'll try this way. For now I filled the missing values using the M421 command (I put a 0.07 value) and the mesh became valid. 

I will try to upgrade the firmware to the 2.1.x bugfix version. Next week I'll be away from home so I'll give you a feedback the next-next week.

I have another problem with the bed but I'm quite sure it is not related to your firmware because I've had it for months. I'm currently in contact with the Creality support but to be honest I'm not very confident that they will found a solution. Basically the mesh created does not compensate correctly for the bed. If you want I can explain to you the issue further deep and maybe you can help me find a solution. This way I'll be able to give you a feedback on the entire bed levelling procedure and even other features because as now I can't print much.

synman, 2023-02-12:
My gut is telling me you got bit by the bad release I had early on.  If so, I've made quite a few additions / improvements since then so it's a win win!

The command you really want is `G29 P1 C`.   It'll probe those unfilled points for ya automagically once you grab the latest version available.

I'll keep this issue here waiting for your next update.  I hope you have a safe trip in the meantime.

synman, 2023-02-28:
closing due to lack of activity