synman/Ender-3-S1-Pro-Firmware #32: UBL 100 Fails at point 28 - F4-s1pro

MrDB42, 2023-02-20:
When running bed leveling, the probe retracts at "Probing mesh point 28/100" and an error is thrown:

Recv: //action:notification Probing Failed
Recv: Error:Probing Failed
Changing monitoring state from "Operational" to "Error"

Happens with either the "bugfix-2.1.x-F4-s1pro_ubl100" or "2.1.2-ender-3-s1-F4-s1pro_ubl100" versions.

MrDB42, 2023-02-20:
Oh, just figured out that it's only when connected via Octoprint. Octoprint will halt the system.

synman, 2023-02-21:
Tweak Octoprint to not fail / halt on all errors.

MrDB42, 2023-02-21:
Oh... That's _cheating_! ;)

I was starting to wonder if my probe was the issue. But if you're running into the same problem, I wonder what the root cause of the errors are?

MrDB42, 2023-02-21:
And... even if I disconnect via Octoprint, running the bed level from the LCD screen, it will eventually lock up completely.

It gets to 100%, then it starts over and hangs at 3%. Weird.

The UBL25 works just fine, BTW.

synman, 2023-02-21:
Have you read the UBL discussion?  Check it out if not as it gives a ton of pointers.

My guess is it's the CR touch flaking out and / or an upstream bug in Marlin.

You're going to find a couple scenarios where Marlin will issue an error where you don't want Octoprint to disconnect.  There are a couple options there you'll want to look at.

synman, 2023-02-21:
Let's move this to the UBL discussion thread

closed 2023-02-21 by synman