synman/Ender-3-S1-Pro-Firmware #33: Request "M425 - Backlash compensation" enabled

MrDB42, 2023-02-20:
I would like to request that "M425 - Backlash compensation" to be enabled on X, Y, and Z axes.

I do mainly "engineering" type of prints and would like to be as accurate as possible on my prints. Backlash compensation would be a HUGE benefit for me. I have tuned my axes as best as possible with a dial indicator, but the lack of backlash compensation is frustrating, for me.

synman, 2023-02-22:
I have this currently enabled in an internal build and will roll it into the next release assuming I don't encounter any showstoppers.

MrDB42, 2023-02-22:
OMG! Thank YOU!!

I was just starting to setup a fork and try to figure out how to do my own build. (It's been a LONG time since I done any coding)

synman, 2023-02-22:
I also require some further input here as there are multiple settings involved, depending on far you take it:

`M425` is but one component of Backlash compensation.   Technically you can have the capability enabled but not include `M425` (`BACKLASH_GCODE`).  This seems stupid to me because you'd end up with firmware hardcoded values so it was a natural leap for me to also enable `M425`.

Things got a little more tricky as a went further into the options available to the capability.  Namely, `CALIBRATION_GCODE` which further extends `M425` but has some specific hardware requirements (see below).  I do not believe this is a wise setting for us.
```
//
// Backlash Compensation
// Adds extra movement to axes on direction-changes to account for backlash.
//
#define BACKLASH_COMPENSATION
#if ENABLED(BACKLASH_COMPENSATION)
  // Define values for backlash distance and correction.
  // If BACKLASH_GCODE is enabled these values are the defaults.
  #define BACKLASH_DISTANCE_MM { 0, 0, 0 } // (linear=mm, rotational=°) One value for each linear axis
  #define BACKLASH_CORRECTION    0.0       // 0.0 = no correction; 1.0 = full correction

  // Add steps for motor direction changes on CORE kinematics
  //#define CORE_BACKLASH

  // Set BACKLASH_SMOOTHING_MM to spread backlash correction over multiple segments
  // to reduce print artifacts. (Enabling this is costly in memory and computation!)
  //#define BACKLASH_SMOOTHING_MM 3 // (mm)

  // Add runtime configuration and tuning of backlash values (M425)
  #define BACKLASH_GCODE

  #if ENABLED(BACKLASH_GCODE)
    // Measure the Z backlash when probing (G29) and set with "M425 Z"
    #define MEASURE_BACKLASH_WHEN_PROBING

    #if ENABLED(MEASURE_BACKLASH_WHEN_PROBING)
      // When measuring, the probe will move up to BACKLASH_MEASUREMENT_LIMIT
      // mm away from point of contact in BACKLASH_MEASUREMENT_RESOLUTION
      // increments while checking for the contact to be broken.
      #define BACKLASH_MEASUREMENT_LIMIT       0.5   // (mm)
      #define BACKLASH_MEASUREMENT_RESOLUTION  0.005 // (mm)
      #define BACKLASH_MEASUREMENT_FEEDRATE    Z_PROBE_FEEDRATE_SLOW // (mm/min)
    #endif
  #endif
#endif

/**
 * Automatic backlash, position, and hotend offset calibration
 *
 * Enable G425 to run automatic calibration using an electrically-
 * conductive cube, bolt, or washer mounted on the bed.
 *
 * G425 uses the probe to touch the top and sides of the calibration object
 * on the bed and measures and/or correct positional offsets, axis backlash
 * and hotend offsets.
 *
 * Note: HOTEND_OFFSET and CALIBRATION_OBJECT_CENTER must be set to within
 *       ±5mm of true values for G425 to succeed.
 */
#define CALIBRATION_GCODE
#if ENABLED(CALIBRATION_GCODE)

  //#define CALIBRATION_SCRIPT_PRE  "M117 Starting Auto-Calibration\nT0\nG28\nG12\nM117 Calibrating..."
  //#define CALIBRATION_SCRIPT_POST "M500\nM117 Calibration data saved"

  #define CALIBRATION_MEASUREMENT_RESOLUTION     0.01 // mm

  #define CALIBRATION_FEEDRATE_SLOW             60    // mm/min
  #define CALIBRATION_FEEDRATE_FAST           1200    // mm/min
  #define CALIBRATION_FEEDRATE_TRAVEL         3000    // mm/min

  // The following parameters refer to the conical section of the nozzle tip.
  #define CALIBRATION_NOZZLE_TIP_HEIGHT          1.0  // mm
  #define CALIBRATION_NOZZLE_OUTER_DIAMETER      2.0  // mm

  // Uncomment to enable reporting (required for "G425 V", but consumes PROGMEM).
  //#define CALIBRATION_REPORTING

  // The true location and dimension the cube/bolt/washer on the bed.
  #define CALIBRATION_OBJECT_CENTER     { 264.0, -22.0,  -2.0 } // mm
  #define CALIBRATION_OBJECT_DIMENSIONS {  10.0,  10.0,  10.0 } // mm

  // Comment out any sides which are unreachable by the probe. For best
  // auto-calibration results, all sides must be reachable.
  #define CALIBRATION_MEASURE_RIGHT
  #define CALIBRATION_MEASURE_FRONT
  #define CALIBRATION_MEASURE_LEFT
  #define CALIBRATION_MEASURE_BACK

  //#define CALIBRATION_MEASURE_IMIN
  //#define CALIBRATION_MEASURE_IMAX
  //#define CALIBRATION_MEASURE_JMIN
  //#define CALIBRATION_MEASURE_JMAX
  //#define CALIBRATION_MEASURE_KMIN
  //#define CALIBRATION_MEASURE_KMAX
  //#define CALIBRATION_MEASURE_UMIN
  //#define CALIBRATION_MEASURE_UMAX
  //#define CALIBRATION_MEASURE_VMIN
  //#define CALIBRATION_MEASURE_VMAX
  //#define CALIBRATION_MEASURE_WMIN
  //#define CALIBRATION_MEASURE_WMAX

  // Probing at the exact top center only works if the center is flat. If
  // probing on a screwhead or hollow washer, probe near the edges.
  //#define CALIBRATION_MEASURE_AT_TOP_EDGES

  // Define the pin to read during calibration
  #ifndef CALIBRATION_PIN
    //#define CALIBRATION_PIN -1            // Define here to override the default pin
    #define CALIBRATION_PIN_INVERTING false // Set to true to invert the custom pin
    //#define CALIBRATION_PIN_PULLDOWN
    #define CALIBRATION_PIN_PULLUP
  #endif
#endif
```

synman, 2023-02-22:
FWIW, I'm looking forward to playing with this a little myself with only `BACKLASH_COMPENSATION`, `BACKLASH_GCODE`, and `MEASURE_BACKLASH_WHEN_PROBING` enabled.

MrDB42, 2023-02-22:
Yeah.... It's a rabbit hole with the autocalibration. Which I don't even now if it's possible with the S1/Pro. You would have to have some conductivity from the nozzle to ground. Which I don't think is there.

But you can use the probe to auto calibrate Z, which is great and I think is turned on from Creality (https://github.com/CrealityOfficial/Ender-3S1/blob/s1_pro/Marlin/Configuration_adv.h). But having the ability to just enter values for X, Y would be a godsend

synman, 2023-02-22:
Yup, I'm with ya.  Good to know you are not looking to enable that final option as I would have to spin up a new variant to support it.

I'll be enabling 3 options listed in the update right above yours.

synman, 2023-02-22:
> But you can use the probe to auto calibrate Z, which is great and I think is turned on from Creality (https://github.com/CrealityOfficial/Ender-3S1/blob/s1_pro/Marlin/Configuration_adv.h). But having the ability to just enter values for X, Y would be a godsend

nah, they have it all turned off.... the entire section is disabled.

https://github.com/CrealityOfficial/Ender-3S1/blob/s1_pro/Marlin/Configuration_adv.h#L975

MrDB42, 2023-02-22:
Ugh. I was wondering about that. It's like that they started, but then quit.

synman, 2023-02-22:
anyway, I think I have what I need to proceed at this point.  Expect to see something firmware wise in the next couple days.  there's a bug related to probing while printing from SD I want to fix too and will bundle these together.

synman, 2023-02-23:
here you go:  https://github.com/synman/Ender-3-S1-Pro-Firmware/tree/main/bugfix-2.1.x

Will leave this open for a couple days to collect feedback / etc.

MrDB42, 2023-02-23:
Just want to confirm that you did not enable "MEASURE_BACKLASH_WHEN_PROBING" for the Z-axis.

I still have to do that manually.

synman, 2023-02-23:
`BACKLASH_GCODE` and `MEASURE_BACKLASH_WHEN_PROBING` are enabled

synman, 2023-02-23:
and I went with the defaults for `MEASURE_BACKLASH_WHEN_PROBING`:
```
      #define BACKLASH_MEASUREMENT_LIMIT       0.5   // (mm)
      #define BACKLASH_MEASUREMENT_RESOLUTION  0.005 // (mm)
      #define BACKLASH_MEASUREMENT_FEEDRATE    Z_PROBE_FEEDRATE_SLOW // (mm/min)
```

MrDB42, 2023-02-24:
That's what I noticed when setting the backlash, 5 micron increments.
On then next go-round, could you set it to 1 micron?
`#define BACKLASH_MEASUREMENT_RESOLUTION  0.001 // (mm)`
When you use a good analog dial indicator, it's easy to "guestimate" 1 micron measurements. I just want to see if it actually makes a difference. (This doesn't work with an electronic dial indicator, because most of them only show hundreds of a millimeter)

And on a that note, for grins-and-giggles, could you also set:
`SERIAL_FLOAT_PRECISION` to 3 or 4 in the advanced configuration section?

synman, 2023-02-24:
Sure thing.  Will possibly get to it tonight.

MrDB42, 2023-02-24:
Take your time. I'm fighting with the new version of Octopi and how it broke my camera setup.

MrDB42, 2023-02-24:
I didn't think it would affect my camera (Aruducam 2.0 electronic/manual focus). So now I can't use the Arducam focusing plug-in.
I've been meaning to kick the mjpeg-streamer. Just hadn't because I eventually got it working perfectly.

Oh, to top it off, I accidentally ripped out the ribbon cable going to my Bigtree touch display trying to get the microSD card out. Trying to reconnect it fsck'd the connector on the display so I had to order a new one. (It was NOT a good day for me).

synman, 2023-02-24:
There are a bunch of "breaking" changes in the latest bullseye builds.

Fact is, we've reached a point where the legacy camera apis really need to die and are no longer functional.

synman, 2023-02-25:
Playing around with this tonight with its built-in calculations with probing Z, I decided to run a probe repeatability test to 1.  generate the backlash statistics for Z and 2.  measure its impact.

Probe Repeatability Test - 10 runs (backlash disabled)
```
M48 P10 X100 Y100 V2
...
Recv: Mean: 0.056085 Min: 0.045 Max: 0.063 Range: 0.018
Recv: Standard Deviation: 0.006873
...
Send: M425
Recv: Backlash Correction inactive:
Recv:   Correction Amount/Fade-out:     F0.00 (F1.0 = full, F0.0 = none)
Recv:   Backlash Distance (mm):         X0.00
Recv:  Y0.00
Recv:  Z0.04
Recv:   Average measured backlash (mm): Z0.04
```
So I then load that measured backlash value in and turn on compensation (I prefer quality over accuracy):
```
Send: M425 Z F1 S3
Recv: ok
[...]
Send: M425
Recv: Backlash Correction active:
Recv:   Correction Amount/Fade-out:     F1.00 (F1.0 = full, F0.0 = none)
Recv:   Backlash Distance (mm):         X0.00
Recv:  Y0.00
Recv:  Z0.04
Recv:   Average measured backlash (mm): Z0.04
```
And re-run my probe test:
```
Send: M48 P10 X100 Y100 V2
Recv: M48 Z-Probe Repeatability Test
...
Recv: Mean: 0.051417 Min: 0.049 Max: 0.053 Range: 0.003
Recv: Standard Deviation: 0.001346
```
This feels significant to me.   The range of the tests dropped from `0.018mm` to `0.003` and check out the std deviation:
```
0.006873
0.001346
```
^^^ That is pretty freaking significant.

MrDB42, 2023-02-25:
That's awesome. 

But in reality, for the Z axis, you want accuracy over quality (if you are doing any type of Z hops). It's only on the X, Y axes that you would set quality over accuracy.

You know what would be really cool, is find a way to reorientate the probe temporarily for the X and Y with a temporary fixed stop and have a way to probe the backlash for X and Y.

So... when can I play with it!?!?!?!? :)

synman, 2023-02-25:
I just need to push the commit... built the binaries a couple minutes ago.   I was doing some reading and apparently z backlash can cause layering issues.  I interpreted as a Z axis problem but I also read about it causing gaps and blobs on x/y.

synman, 2023-02-25:
its out there now

synman, 2023-02-25:
I'm going to close this as an issue and move it to Discussions with the Feature label.