synman/Ender-3-S1-Pro-Firmware #34: changing z offset error while printing

dodolcool, 2023-02-21:
Hello i have issue with one of my printer.
around a week ago i tried change my new s1pro to this FW after factory reset and leveling i start print test and manually change z offset using lcd for the first time it stop the print after turnoff and on the printer this issue not occurs anymore.

today i change my old s1pro to this printer and the same behaviour occur again but this time i notice the increase number back to 0.05 not 0.01 and as always the print stuck. if i change the z using leveling mode it can work as espected. already follow your steps match the chip with the FW, factory reset using gcode and using lcd but still can't find the solution

synman, 2023-02-21:
I'm going to have to recreate this scenario myself.  Thanks for opening the ticket.

I had heard through Facebook there was something funky going on with z-offset and SD card printing.

I'm guessing this is it.  The latest bugfix-2.1.x release has some conditional logic if printing from SD that is supposed to use the original Creality logic / code.

I'm gathering it didn't work.

I am not an SD print kinda guy so this isn't something I do as part of normal validation.

dodolcool, 2023-02-22:
ahh yes this issue occurs on machine that print from sdcard. while the other printer connect to octoprint and it work as intended

synman, 2023-02-22:
I'm going to try and sort this out this evening.

synman, 2023-02-23:
So yeah, I am unable to reproduce this.   My only conclusion at this point is you are using an older version of the firmware that did not properly account for modifying Z-Offset while printing from the SD card.

Yes, it is going to increment in .05mm increments.  No, I am not going to try and change this.  If you want finer control over the zoffset, print from an external sender such as Octoprint or pick your sender of choice.

https://user-images.githubusercontent.com/1299716/220800560-e02bf5c9-d125-4ced-8b2d-c1ee70fce509.mp4

I'l alsol be updating the bug template here shortly to remind folks to always test with the latest firmware version in the files section here before opening a bug report.

dodolcool, 2023-02-23:
beforu update to your FW im using Ender-3 S1_Pro_HWv24S1_301_SWV2.0.8.24F4_FDM_LASER 29 Nov. 2022 this FW. if im not wrong you mention to use nov fw instead of newest feb 2023 fw

so this still be mystery for us?

synman, 2023-02-23:
My machine's baseline firmware is the November firmware.  I never flashed the February version myself.

dodolcool, 2023-02-23:
already tried all of them still get the same issue.

maybe today i will tried to install in third mechine to check if this issue also occurs or not

synman, 2023-02-23:
please be very specific if you continue to experience this issue.  I want to help you but I need accurate details, especially since I am unable to recreate this issue myself.

At the very least, capture the filename of the .bin/.zip file you downloaded and when.  I have been hesitant to generate tags / releases so please ensure you have this information.

synman, 2023-02-23:
I'll leave this open for now pending more feedback from you (or others).

dodolcool, 2023-02-24:
[image]
Hello i already tried in 4 different machines all the machines originaly used fw like the image  above.
the condition are:
2 machines connect to octoprint
2 machines using sd card.

step to repro  Octoprint:
1. from FW 2.0.8 24f4 upgrade to 2.1.2PA
2. perform m502 using octoprint
3. restart the printer
4. leveling like other people do (adjust z first then adjust knob then perform auto leveling)
5. perform test print
7. adjust the z while printing rapidly.
8. notice that the print will stop. the issue will not occurs if changeing the Z with some times.

Direct SD CARD Print
1. from FW 2.0.8 24f4 upgrade to 2.1.2PA
2. perform m502 using pronterface
3. restart the printer
4. leveling like other people do (adjust z first then adjust knob then perform auto leveling)
5. perform test print
7. adjust the z while printing just press once
8. notice that the print will stop 

the result:
2 machines that connect to octoprint work as intended. even though if increase/decrease rapidly while printing it will stop the print. but if you give some time between increase/decrease it work fine.

the other 2 machine sill not working even pressing increase/decrease once the print stop. 

conclusion:
Using octoprint this FW work fine but you need to be patien when setting the Z offset while printing
Direct ptint from SD card still get stuck even only press once increase or decrease while printing.

additional information:
-Octoprint machine tested using both 208PA or 212PA and the behaviour same as describe above
-Direct SD Card tested using 208PA, 212PA and 21xBugFix and the behaviour same as describe above
-SD Card used to test 8GB class 10 obtained from purchasing the machine.
-I can't test using differnt SD Card since i don't have spare SD Card.

synman, 2023-02-24:
Use the bugfix-2.1.x builds.  Please note these fixes have not been back ported to the other branches.

At this point I may drop 2.0.8 and 2.1.2 support entirely as it is a PITA keeping 3 disparate versions of Marlin in sync.

It's not like I'm getting paid to do this.  The incentive simply isn't there for me to keep up my initial pace.

closed 2023-02-24 by synman