synman/Ender-3-S1-Pro-Firmware #37: Second print in a row fails

lipa46, 2023-02-26:
[X ] put an X in the box to the left to indicate you have followed the steps mentioned above.  Bug reports that do not have this box marked will be closed and locked.

**Describe the bug**
After a successful completed print, i would like to initialize a second print thru octoprint. the print is initialized but at the start i check for bed changes with command 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Send: N9 M420 L0 V*6
Recv: Mesh loaded from slot 0
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  0,183)                                                              (203,183)
Recv:         0       1       2       3       4       5       6       7       8       9
Recv:  9 | +0.1710  +0.1730  +0.1860  +0.1560  +0.1650  +0.1840  +0.1360  +0.1880  +0.2200  +0.1580
Recv:    |
Recv:  8 | +0.1810  +0.1930  +0.1750  +0.1470  +0.1640  +0.1630  +0.1120  +0.1560  +0.1780  +0.1380
Recv:    |
Recv:  7 | +0.1570  +0.1850  +0.1680  +0.1070  +0.1220  +0.1290  +0.0880  +0.1250  +0.1360  +0.1170
Recv:    |
Recv:  6 | +0.1010  +0.1500  +0.1330  +0.0730  +0.0900  +0.0900  +0.0640  +0.1020  +0.1230  +0.0920
Recv:    |
Recv:  5 | +0.0970  +0.1180  +0.1040  +0.0680  +0.0750  +0.0700  +0.0300  +0.0690  +0.0760  +0.0630
Recv:    |
Recv:  4 | +0.0690  +0.1020  +0.0920  +0.0490  +0.0780  +0.0490  +0.0310  +0.0770  +0.0740  +0.0480
Recv:    |
Recv:  3 | +0.0730  +0.0830  +0.0900  +0.0340  +0.0620  +0.0470  +0.0260  +0.0790  +0.0770  +0.0770
Recv:    |
Recv:  2 | +0.0450  +0.0600  +0.0440  -0.0070  +0.0210  +0.0280  -0.0140  +0.0360  +0.0150  -0.0090
Recv:    |
Recv:  1 | +0.0660  +0.0720  +0.0550  +0.0020  -0.0070  -0.0100  -0.0260  +0.0200  +0.0210  -0.0050
Recv:    |
Recv:  0 | +0.1930  +0.2050  +0.1560  +0.0970  +0.0860  +0.0920  +0.0880  +0.1500  +0.1590  +0.1340
Recv:         0       1       2       3       4       5       6       7       8       9
Recv:     (  0,  0)                                                              (203,  0)
Recv: 
Recv: Mesh is valid
Recv: Storage slot: 0
Recv: echo:Bed Leveling OFF
Recv: echo:Fade Height 10.00
Recv: ok
Send: N10 G29 J2 T*15
[...]
Recv: Tilting mesh point 1/4
Recv: 
Recv: //action:notification Tilting Point 1/4

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Machine (please complete the following information):**
 - S1 Pro / S1 Plus

**Variant (please complete the following information):**
 - UBL 25
 - UBL 100
 - ABL

**Marlin Version (please complete the following information):**
 - bugfix-2.1.x
 - 2.1.2
 - 2.0.8

**Additional context**
Add any other context about the problem here.

synman, 2023-02-26:
I think I stated previously it sounds like you have a hardware issue.  

I have some prints I need to run today to rule out the firmware as I did just release a new version for all variants/branches merged from upstream and back ported (where possible).

synman, 2023-02-26:
Confirmed no firmware issues with latest firmware.  Will be closing this as invalid.

lipa46, 2023-02-26:
Ok I will check the probe for any issues. During the UBL100 point test it fails consistently 3 times, it could be HW. But previously i haven't had any problems. Thanks for your help and the development of this firmware.  I will report the result back.

synman, 2023-02-26:
Even my CR touch fails typically 3 times during the 100 point probe.  **What it has never done is fail right off the bat.**  I held back on releasing a UBL variant until I could understand exactly what is happening.  

It's within the bowels of the Marlin probe.c code where it fails.  Marlin asks for a retract and the CR touch flakes out.  There's not a whole lot of rhyme or reason to it either.  But these conditions you've been describing do not match my own experience, or you have not described them with sufficient detail for me to correlate them to expected behavior.

You have no idea how relieved I was to find Marlin accommodates this type of oddity with the `G29 P1 C` command.

I've been considering an EZABL probe for my S1 Pro as that is what I run on another machine and its general accuracy / repeatability is 1000 times better.   Doing so would require a completely new variant and since it's "good enough" right now, I've held off on pulling that trigger.

If I see something pop via the Marlin upstream I'll be sure to mention something, but as it stands now, this is the best I've got.

There's one other piece of data here that leads to me to think the CR touch itself is just buggy.  I'll share that in a couple minutes as I have to pull some Creality source to demonstrate it.

synman, 2023-02-26:
Check this out:  https://github.com/CrealityOfficial/Ender-3S1/blob/s1_pro/Marlin/Configuration_adv.h#L1947

There's an actual hack for working around this built-in to the stock firmware running ABL.  The UBL authors apparently refused to accept this hack when they developed UBL as SanityChecks will not allow it to be set when UBL is enabled:  https://github.com/CrealityOfficial/Ender-3S1/blob/c9efd59a0aca75468191629dfaffffcc701a0922/Marlin/src/inc/SanityCheck.h#L1572

closed 2023-02-26 by synman