synman/Ender-3-S1-Pro-Firmware #41: Noticed bugs

Zharkoviv, 2023-03-03:
Printer Ender3 s1 plus
bugfix-2.1.x-F4-s1pro_ubl25 firmware-20230218-153855 Normal operation without noticed errors
firmware-20230301-023212 Automatic calibration error (203 Probing Failed ), z-axis correction 0.01 mm, default 0.05.

STM32F401RC_creality_s1pro_ubl25 firmware-20230301-023212 Printing works, incorrect printer model and table size

synman, 2023-03-03:
Where are you getting incorrect printer model and table size?  Please try to type in complete sentences and fully explain what you are experiencing.  Use Google Translate if language is a problem.  I would rather work with a poor translation than a couple lines of text that have no context / meaning.

This is the file file you referenced `firmware-20230301-023212.bin` while viewed in a hex editor:

[image]

synman, 2023-03-03:
I did not want to leave this with only providing a hex dump of the bin file.  I flashed this firmware to my S1 Pro and was able to run a 25 point G29 via the touchscreen without issue.  It even stowed the probe after point 20 like it should.

https://user-images.githubusercontent.com/1299716/222607891-fef337c5-174e-42c7-9995-4ffa693102b9.mp4

I now have a clean 25 point UBL mesh stored in slot 1 and everything on the machine and display state the correct information model / bed size / version / etc.

```
Send: M420 L1 V
Recv: Mesh loaded from slot 1
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  0,183)                      (203,183)
Recv:         0       1       2       3       4
Recv:  4 | +0.0680  -0.0160  -0.0010  -0.0010  +0.0860
Recv:    |
Recv:  3 | +0.1730  +0.0980  +0.0960  +0.0330  +0.0880
Recv:    |
Recv:  2 | +0.1780  +0.0860  +0.0780  +0.0130  +0.0610
Recv:    |
Recv:  1 | +0.1630  +0.0330 [-0.0030] -0.0760  -0.0380
Recv:    |
Recv:  0 | +0.2160  +0.0710  +0.0160  -0.0340  +0.0310
Recv:         0       1       2       3       4
Recv:     (  0,  0)                      (203,  0)
Recv: 
Recv: Mesh is valid
Recv: Storage slot: 1
Recv: echo:Bed Leveling OFF
Recv: echo:Fade Height OFF
Recv: ok
```

Zharkoviv, 2023-03-04:
> 

Sorry in a hurry I downloaded the wrong file. As soon as I try a new version of the firmware, I will try to describe everything in detail.

synman, 2023-03-04:
> > 
> 
> Sorry in a hurry I downloaded the wrong file. As soon as I try a new version of the firmware, I will try to describe everything in detail.

Take a look here and feel free to join in:. https://github.com/synman/Ender-3-S1-Pro-Firmware/discussions/38#discussioncomment-5202998

I suspect we are chasing the same issue.

synman, 2023-03-04:
And here where someone was able to tune their machine to address this: https://github.com/synman/Ender-3-S1-Pro-Firmware/discussions/16#discussioncomment-5188078

synman, 2023-03-04:
I'm going to close this because it doesn't speak to the actual issues being researched.  Rest assured, I'm looking to figure out what is going on with a subset of machines experiencing this probe failed condition.

closed 2023-03-04 by synman