synman/Ender-3-S1-Pro-Firmware #42: Z Offset Raise Limitation.

AnthonyUDT, 2023-03-04:
Whenever i try to raise my zoffset above 5.0 i am unable to due to the limitation of the firmware. 
Reason being i want to use the E-Leveler and it requires the firmware zoffset to be 5.8mm to compensate.
If you could unlock the limit of the zoffset or describe how i could change and compile myself.

https://user-images.githubusercontent.com/25626288/222927456-379a591b-6fa1-48e6-be56-0b35513cd04e.MOV

synman, 2023-03-04:
I can address this fairly easily.  Will be a day or two.

synman, 2023-03-04:
I'm going to need a few more details.   Which variant / build are you using?  My initial review indicates the min/max are -10,10 across the board, at least within the bugfix-2.1.x builds.

AnthonyUDT, 2023-03-04:
Hello these are the firmware versions i have tested different printers
[image]
[image]

synman, 2023-03-04:
Please share the actual filename for the .bin file you are flashing.  Even better, just grab the latest version and always ensure you are testing with the latest version before filing an issue.

The probe offset increment should be .01mm when accessed via the aux menu (assuming you do not have a Gcode file loaded via the sdcard).   

FWIW, I did confirm something is preventing the touchscreen from going beyond +5mm and will get to the bottom of it.

synman, 2023-03-04:
For reference :)

https://user-images.githubusercontent.com/1299716/222929841-c7024902-7e69-49e1-90ba-e635aa402047.mp4

AnthonyUDT, 2023-03-04:
i apologize for not clarify, 
1. firmware-20230225-131954.bin (FW MD5 83b4c8cccedf2c70906562801cc98cf9)
zip name (bugfix-2.1.x-F4-s1pro_abl.zip)
2.  firmware-20230209-013023.bin (FW MD5 c6ea30d8eea45ff725dd311e62b8482e)
zip name (2.1.2-ender-3-s1-F4-s1pro_abl.zip)

synman, 2023-03-04:
yup, np.... always test with the latest and greatest.... you get a pass here though cause it's broke even in the latest build!

synman, 2023-03-04:
@AnthonyUDT I'm curious .... are you familiar enough with the stock firmware to confirm whether or not you were able to go above 5mm offset on it?   I'm not absolutely certain yet but the touchscreen may actually be blocking / undoing this.

synman, 2023-03-04:
Case in point, I'm able to push right past 5mm when executing `M290` directly:

```
Send: M851
Recv:   M851 Probe Offset X-31.5000 Y-41.8000 Z4.9900 ; (mm)
Recv: ok
Send: M290 Z+.01
Recv: echo:Probe Offset Z5.0000
Recv: ok
[...]
Send: M290 Z+.01
Recv: echo:Probe Offset Z5.0100
Recv: ok
[...]
Send: M290 Z+.01
Recv: echo:Probe Offset Z5.0200
Recv: ok
[...]
Send: M290 Z+.01
Recv: echo:Probe Offset Z5.0300
Recv: ok
```

synman, 2023-03-04:
I think I've got what I need at this point.... this is going to be a `cantfix` for now because the touchscreen is hardcoded to max out a 5.00mm.  I've confirmed the by sniffing the underlying data it sends and once I go beyond 5mm (+ or -) it tops out at that value.

And to make matters worse, since all we have to work with is an "enter key" action I have no way to determining whether the positive or negative button was pushed because the offset the touchscreen returns is capped at this value.

synman, 2023-03-04:
Lastly, I bet the stock firmware has the same problem.   We may be able to hack the touchscreen binary to increase this limit but that is a whole other exercise.

If you find newer versions of the touchscreen do not have this limitation, please let me know.  As of now, I'm 100% certain we are DOA with the November one.

AnthonyUDT, 2023-03-05:
Alright, thanks you for checking. Also do you have a community channel or discord/telegram server ? I am interested in the development of the firmware and want to be notified of future updates. I wouldn't also mind contributing.

closed 2023-03-05 by AnthonyUDT

synman, 2023-03-05:
I'm more than happy to establish a discord server if there's any interest.  I frequent a number of different discord servers already.