synman/Ender-3-S1-Pro-Firmware #44: Cannot continue printing after a pause

synman, 2023-03-05:
### Discussed in https://github.com/synman/Ender-3-S1-Pro-Firmware/discussions/40

<div type='discussions-op-text'>

<sup>Originally posted by **urdigor13** March  2, 2023</sup>
Unable to continue printing after g-code M25. When you press the "continue printing" button on the display, nothing happens, printing does not continue. I also tried to pause using the display. It also fails to continue printing. Tell me what could be the problem?</div>

urdigor13, 2023-03-05:
I tried to send the m24 command via pronterface, but printing also does not continue.

urdigor13, 2023-03-05:
@synman, can you tell me why the m25 command causes the extruder to park? Shouldn't he stay put? At least in stock firmware the m25 command causes the extruder to freeze in place.

synman, 2023-03-05:
Advanced Pause

synman, 2023-03-05:
Did you notice the slew of host action prompts that are accompanying the pause?  

There are quite a few variations of pause, resume, cancel here I'm working to integrate.

It will hopefully all make more sense once I deliver the functionality in its entirety.

Mikerhinos, 2023-03-25:
It looks like printing does not resume after a M600 change filament gcode too :(

bvweerd, 2023-03-25:
Seems to be a known issue in Marlin, fix is also described here: https://github.com/MarlinFirmware/Marlin/issues/24996

Mikerhinos, 2023-03-25:
Yes thanks I was reading it too, so at the moment there is no way to bypass this and change filament while printing ?

bvweerd, 2023-03-25:
Not without implementing the fix first...

synman, 2023-03-25:
I doubt these are related.  I had validated filament change via the runout sensor some time back when using an external sender such as Octoprint exclusively.

I'm taking a break right now but will be returning here at some point in the not too distant future.

Mikerhinos, 2023-03-26:
May be related to their proprietary screen : I flashed official firmware (latest from february) and filament change is working BUT to resume printing you have to go in "live settings adjustment" (upper part of screen), then go back so the pause icon changes to a play icon, then you can click it and choose "continue"...

urdigor13, 2023-03-30:
Hello Mikerhinos! I found another firmware build for Ender 3 s1 pro. It contains everything I need: Linear Advance, text output to the display works through the g-code M117. And also the pause (M25) works adequately in it. Try it, most likely there and the M600 will work fine.
https://github.com/ThomasToka/MarlinFirmware/

Mikerhinos, 2023-03-31:
> Hello Mikerhinos! I found another firmware build for Ender 3 s1 pro. It contains everything I need: Linear Advance, text output to the display works through the g-code M117. And also the pause (M25) [...]

Thanks ! I'll try to check it out this week-end.