synman/Ender-3-S1-Pro-Firmware #46: Ender 3S1 Plus with touch screen issues a confirmation but without one on the screen

pronchakov, 2023-03-09:
**STOP - DO NOT PROCEED - YET**

I know you are itching to get something fixed and/or you really need help with a problem you are experiencing but before you submit this report, go to the files section and download the latest version of the firmware for your machine.

If you can recreate the issue using the latest available version of the firmware, come back here and feel free to proceed with creating a bug report.

**FIRST STEP**

[X] put an X in the box to the left to indicate you have followed the steps mentioned above.  Bug reports that do not have this box marked will be closed and locked.

**Describe the bug**
I am using build from commit ed89a2d552ef80124308e6287a927af8505d7653.
bugfix-2.1.x-F4-s1plus_abl.zip
Using PrusaSlicer 2.5.0
Start Gcode:
M104 S150
M190 S{first_layer_bed_temperature[0]}
G28
M109 S{first_layer_temperature[0]}

But after extruder get heated to it's working temperature(M109), printer wont go further. 
If I connect using pronterface, it's sending "echo:busy: paused for user" message ocer and over.
Only after M108 command from pronterface print is proceeding to print.

**To Reproduce**
Steps to reproduce the behavior:
1. Slice test figure using start gcode above
2. Start printing
3. After extruder get heat, see error in pronterface console
4. Use M108 to continue

**Expected behavior**
No M108 commad should be needed for print to start

**Screenshots**
None

**Machine (please complete the following information):**
 - S1 Plus

**Variant (please complete the following information):**
 - ABL

**Marlin Version (please complete the following information):**
 - bugfix-2.1.x

**Additional context**
None

pronchakov, 2023-03-09:
Just tried put M108 at the end of Start Gcode but with no luck

pronchakov, 2023-03-10:
A little update.
If I communicate with printer using touch screen while initial M109 is waiting, for example, set another temperature, pause(waiting for user input) is never occur.

pronchakov, 2023-03-10:
Hmm, but not every time that trick works.

pronchakov, 2023-03-10:
Update: pause occurs only with pronterface connected to printer.

Torallulu, 2023-03-13:
hi to all i didnt understand the steps for flashing this firmware to plus model. is the bed dimensions are set to 300x300 or is set for pro models bed (220x220)
i m not very well on firmware moding can someone help with the steps
When i was using a cr6 i both flashed board and screen seperately with community firmware but when i download this packages
there are 3 zip files including bin files in them which one is for screen which one is for board and what is third one for?
Sorry for questions but meaby my knowledge is not very well to solve this
[image]
in this installation plan it only says insert your sd to printer what about screen microsd? Which one of these should be on the root or wherever it should be

[image]
what are all these bins for there are 6 of them i think if im not wrong 3 of them are compatible with my s1 plus but why 3 and where should i put them thanks for your time and regards..

shardsofaperture, 2023-03-26:
Watching this because I also have an s1 plus and would like to try this firmware but had the same question regarding bedsize and what to do when I go to flash.