synman/Ender-3-S1-Pro-Firmware #50: Unable to run automatic bed leveling

RedFoxy, 2023-04-13:
After upgrading to 2.1.2 firmware (ABL/UBL25/UBL100) I can't use the bed leveling procedure, I can do the manual one but not the automatic

**To Reproduce**
Just run the automatic procedure of Bed Leveling

**Expected behavior**
After changing the firmware, following your guide, using ABL firmware the automatic bed leveling it doesn't start and give "203 (Probing Failed)" error, using UBL25 or UBL100 firmware it starts but crash/stops to works with fan at maximum speed when it goes in top left of the bed.

Before to starts I downgrade the firmware to Ender-3s1pro_hw24s1_301_V2.0.8.16F1_F103_LASER_FDM.zip
EVERYTIME I changed the firmware, I do a factory reset from the display and run M502, M500 and M501 from Octoprint terminal.

[image]

**Machine (please complete the following information):**
Creality Ender 3 S1 Pro bought 3 april 2023 F/W Ver 2.0.8.24F4

[image]

**Variant (please complete the following information):**
 - UBL 25
 - UBL 100
 - ABL

**Marlin Version (please complete the following information):**
 - 2.1.2

avicouzin, 2023-04-16:
I got the same issue. Same printer and firmware

PsyRox2020, 2023-04-19:
Me too, same issues, same printer and firmware. BTW: Do i need anything else than this firmware. Is there any tutorial oder more detailed description availible?

mlebaugh, 2023-04-20:
I was having the same issue on a printer I just picked up. I have 4 of these total, and the other 3 were running this firmware with no issues. 

I used the source to compile with Z_PROBE_LOW_POINT at -10 per the marlin troubleshooting. https://marlinfw.org/docs/basics/troubleshooting.html#g29-probing-failed! 

I then updated the firmware and ran bed leveling from octoprint via the bed visualizer plugin, all of the points were within .2mm so I have no idea why probing failed.

Robertwhitten87, 2023-04-21:
> I was having the same issue on a printer I just picked up. I have 4 of ender 3 s1 pros, and the other 3 were running this firmware with no issues.
> 
> I used the source to compile with Z_PROBE_LOW_POINT at -10 per the marlin troubleshooting. https://marlinfw.org/docs/basics/troubleshooting.html#g29-probing-failed!
> 
> I then updated the firmware and ran bed leveling from octoprint via the bed visualizer plugin, all of the points were within .2mm so I have no idea why probing failed.

Yeah, it's a very perplexing issue. So with just fiddling around and remembering a problem I had with a BTT board and Klipper. I played with the wiring and if I held the wiring into the socket, it would successfully finish the mesh. Which makes me wonder if there's some kind of noise or connection issue. Though with stock firmware it also completes just fine.

mlebaugh, 2023-04-21:
I didn’t change anything except the firmware setting for z probe low point, so I doubt its wiring or noise. Im just not sure why that setting fixed it, since the bed wasn’t out of level more than a few tenths of a mm.

Robertwhitten87, 2023-04-24:
> I didn’t change anything except the firmware setting for z probe low point, so I doubt its wiring or noise. Im just not sure why that setting fixed it, since the bed wasn’t out of level more than a few tenths of a mm.

I've attempted to follow the same suite, and man I can't seem to get a compiled version to work. Mind if I message you?

kemo87, 2023-04-27:
Same issue here. New S1 Pro, bought this week.

Auxx, 2023-05-06:
Same issue, here's the log from OctoPrint:

```
Recv: Default storage slot 0 selected.
Recv: Mesh invalidated. Probing mesh.
Recv:  T:31.79 /0.00 (3846.8125) B:50.03 /50.00 (3611.5000) @:0 B@:12
Recv: echo:busy: processing
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  0,183)                                                              (203,183)
Recv:         0       1       2       3       4       5       6       7       8       9
Recv:  9 |   .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv:  8 |   .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv:  7 |   .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv:  6 |   .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv:  5 |   .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv:  4 |   .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv:  3 |   .       .       .    [  .   ]   .       .       .       .       .       .
Recv:    |
Recv:  2 |   .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv:  1 |   .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv:  0 |   .       .       .       .       .       .       .       .       .       .
Recv:         0       1       2       3       4       5       6       7       8       9
Recv:     (  0,  0)                                                              (203,  0)
Recv: 
Recv: Probing mesh point 1/100.
Recv:  T:31.05 /0.00 (3853.9375) B:49.56 /50.00 (3618.6875) @:0 B@:52
Recv: echo:busy: processing
Recv:  T:31.88 /0.00 (3845.9375) B:49.99 /50.00 (3612.1875) @:0 B@:0
Recv: echo:busy: processing
Recv:  T:31.79 /0.00 (3846.8125) B:50.01 /50.00 (3611.7500) @:0 B@:12
Recv: echo:busy: processing
Recv:  T:31.77 /0.00 (3847.0000) B:49.98 /50.00 (3612.3750) @:0 B@:16
Recv: echo:busy: processing
Recv:  T:31.78 /0.00 (3846.9375) B:49.95 /50.00 (3612.8125) @:0 B@:19
Recv: Error:Probing Failed
Changing monitoring state from "Operational" to "Error"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor
```

shardsofaperture, 2023-05-07:
Is your fast probe disabled?

Run M401 S0 to disable fast probe the. M500 to save to eeprom and try auto level again.

RedFoxy, 2023-05-08:
I sent it by octoprint

shardsofaperture, 2023-05-08:
> I sent it by octoprint

The turn off fast probe command and then saved it? Did it resolve the issue when trying to rerun bed leveling?

Nosnahc, 2023-05-09:
> > I sent it by octoprint
> 
> The turn off fast probe command and then saved it? Did it resolve the issue when trying to rerun bed leveling?

Didn't solve the problem no.

RedFoxy, 2023-05-09:
> > > I sent it by octoprint
> > 
> > 
> > The turn off fast probe command and then saved it? Did it resolve the issue when trying to rerun bed leveling?
> 
> Didn't solve the problem no.

+1

Nosnahc, 2023-05-09:
I have the 1.0.3 Screen firmware.
Does I have to downgrade it?

mlebaugh, 2023-05-10:
> Is your fast probe disabled?
> 
> Run M401 S0 to disable fast probe then M500 to save to eeprom and try auto level again.
> 
> I assume you know how to do this but if not running octoprint or something else to send gcode commands, just put both commands in a text file, save as whatverfilename.gcode to your SD card, and run that file on your touch screen then try abl/ubl again. That's how I got my UBL25 to work on my plus without probe fail errors.

This resolved the issue for me.

Nosnahc, 2023-05-10:
> This resolved the issue for me.

Which printer?
Which firmware and screen firmware version please ?

RedFoxy, 2023-05-10:
> Which printer? Which firmware and screen firmware version please ?

Ender-3 S1 Pro  3 april 2023

ftj2006, 2023-05-11:
> Ender-3 S1 Pro 3 april 2023

Is it v1.0.2 or 1.0.3?

From what I've seen, the readme does not call out a specific screen firmware version(although v1.0.2 is in the screenshot) just the actual printer firmware version.

I have yet to try the fast probe gcode settings to see if that resolves the issue but if not I'll try downgrade to v1.0.2 on the screen

Nosnahc, 2023-05-11:
> Is it v1.0.2 or 1.0.3?
>
> From what I've seen, the readme does not call out a specific screen firmware version(although v1.0.2 is in the screenshot) just the actual printer firmware version.
>
> I have [...]

I'm on 1.0.3 too, but they are on Ender 3-S1 Pro when I'm on a Plus version.

The 2.0.8.25 firmware is delivered with a 1.0.3 touchscreen firmware I think

ftj2006, 2023-05-11:
> I'm on 1.0.3 too, but they are on Ender 3-S1 Pro when I'm on a Plus version.
>
> The 2.0.8.25 firmware is delivered with a 1.0.3 touchscreen firmware I think

I also have the plus and same issues, I am curious if 1.0.2 screen would fix it, will test it out over the weekend

Nosnahc, 2023-05-11:
I will try to downgrade to the end of my current print.
But I believe that the readme version gives a touchscreen firmware at 1.0.3

RedFoxy, 2023-05-11:
> I will try to downgrade to the end of my current print. But I believe that the readme version gives a touchscreen firmware at 1.0.3

but the touchscreen version isn't connected to the firmware version, it's only a grafical view of the information, on the mainboard of the Ender 3 S1 Pro there is still a connector for the old display with the wheel, you can disconnect the display and send commands by usb

Nosnahc, 2023-05-15:
> but the touchscreen version isn't connected to the firmware version, it's only a grafical view of the information, on the mainboard of the Ender 3 S1 Pro there is still a connector for the old [...]

Then why we can read:
 "S1 Plus
Make sure you are running v2.0.8.25 of the TouchPanel Display and the Machine before proceeding any further."
 in the readme?

Nosnahc, 2023-05-16:
Resolve my problem using [this firmware](https://github.com/ThomasToka/MarlinFirmware).
Frequently updated and working great on my Ender 3-S1 Plus.

fernplant, 2023-08-29:
> Resolve my problem using [this firmware](https://github.com/ThomasToka/MarlinFirmware). Frequently updated and working great on my Ender 3-S1 Plus.

I gave this a shot and my Ender 3 S1 Pro and it failed to load the firmware, just hung on black screen. I reverted back to Stock Creality firmware and its back running. Shame these dont seem to work