synman/Ender-3-S1-Pro-Firmware #63: New Firmware based on Creality Firmware version Ender-3 S1_Pro_HWv24S1_301_SWV2.0.8.28F4_F401_FDM_LASER

DDdotDE, 2023-10-15:
**Is your feature request related to a problem? Please describe.**
Creality released New Firmware vor the F4 version of the ender 3 s1 pro and plus, aswell as new firmware for the display
adding the ability to view the mesh and editing it.
at the moment this is not compatible with the custum firmware of yours(if i click the edit leveling button nothing happens)

**Describe the solution you'd like**
id like to see compatibility with the new display firmware

**Additional context**
you can find the new firmware here under ender3 s1 pro / plus then 
Ender-3 S1_Pro_HWv24S1_301_SWV2.0.8.28F4_F401_FDM_LASER
(its from august and it seems like the F1 version is not supportrt)
https://www.crealitycloud.com/software-firmware/firmware/ender-series
as far as i know the source code is not currently available, so i suspect that this isn't currently doable.

synman, 2023-10-28:
I have just been running with my last build from what feels like ages ago.  You have peaked my curiosity though and I'll look into this further at some point, assuming Reality also released the source code like they did previously.

I'm more than happy to swing back in with all the knowledge I gained porting this previously to bring their latest monstrosity of a firmware to a modern version of Marlin, if required.

synman, 2024-03-08:
I haven't had the cycles to devote any time to my ender firmware project for quite some time.   It is safe for you  to assume anything outstanding is not prioritized as nothing outstanding has gotten in the way of any of my own 3d printing needs.

I'm sure this isn't what you want to hear / read, but I also do not want to set a false expectation that I am still actively maintaining this project.

I believe I've got the latest screen firmware documented that I know for a fact is supported.  I also recall there being a screen update (mainly focused on enhancements for laser support) Creality released that folks here were able to verify worked without issue.

DDdotDE, 2024-03-09:
Hey,
So i actuly found another (more actively maintaind) Firmware project, shortly after I opend thos issue.

[Its this on by ThomasToka](https://github.com/ThomasToka/MarlinFirmware)

I would suggest to check it out, i hope you are okay with that synman.

P.s. pay attention to the type of  hotend, the S1 Pro has a different than the standart ender 3's. Talking out of experiance.

Shreyas-ITB, 2025-02-27:
Yo guys, my printer (Ender 3 S1 Pro) is currently running 2.0.8.28F4_F401 firmware (I just updated it from 2.0.8.26F4 which was stock that came when I bought the printer recently) I flashed the new firmware because I wanted to check if they had the Host Action Commands enabled so that I could use OctoPrint with my printer. But unfortunately, no they have not enabled it (I saw a post on reddit which was dated 3 years old convo about creality enabling Host Action Commands on their latest firmware related to the S1 Pro, so i had to check out). I don't want to roll back to old firmware because there can be bugs or I'm very concerned about bricking my printer making it unusable at some point. I just wanted to ask if @synman could help us out with the new firmware release that has the Host Action Commands enabled 😄Thanks a lot!